ArmDeveloperEcosystem/microphone-library-for-pico #1: Compile errors: not compatible with TinyUSB 0.10?

gigamegawatts, 2021-07-04:
Hi,

When building the project I get a lot of errors related to TinyUSB -- see below.

Version 1.2 of the Pico SDK updated TinyUSB from 0.8.0 to 0.10.1, and the release notes mention:
"Note also that moving from TinyUSB 0.8.0 to TinyUSB 0.10.1 may require some minor changes to your USB code."

I'm guessing that these errors are caused by the TinyUSB update, and changes are required tusb_config.h and usb_descriptors.c -- perhaps main.c as well.  I noticed that the TinyUSB audio_test example has quite a few changes to these files.

Can someone please confirm that this project isn't compatible with TinyUSB 0.10.1?

Thanks,

Dan

_Errors from build of usb_microphone:_

>[ 69%] Building C object examples/usb_microphone/CMakeFiles/usb_microphone.dir/main.c.obj
In file included from C:\Users\gigam\Pico\pico-sdk\lib\tinyusb\src/tusb.h:80,
                 from C:\Users\gigam\source\repos\pico\pico-microphone-v1\examples\usb_microphone\usb_microphone.h:11,
                 from C:\Users\gigam\source\repos\pico\pico-microphone-v1\examples\usb_microphone\main.c:17:
C:\Users\gigam\Pico\pico-sdk\lib\tinyusb\src/class/audio/audio_device.h:40:2: error: #error You must tell the driver the length of the audio function descriptor including IAD descriptor
   40 | #error You must tell the driver the length of the audio function descriptor including IAD descriptor
      |  ^~~~~
In file included from C:\Users\gigam\Pico\pico-sdk\lib\tinyusb\src/tusb.h:80,
                 from C:\Users\gigam\source\repos\pico\pico-microphone-v1\examples\usb_microphone\usb_microphone.h:11,
                 from C:\Users\gigam\source\repos\pico\pico-microphone-v1\examples\usb_microphone\main.c:17:
C:\Users\gigam\Pico\pico-sdk\lib\tinyusb\src/class/audio/audio_device.h:55:2: error: #error You must tell the driver the number of Standard AS Interface Descriptors you have defined in the audio function descriptor!
   55 | #error You must tell the driver the number of Standard AS Interface Descriptors you have defined in the audio function descriptor!
      |  ^~~~~
C:\Users\gigam\Pico\pico-sdk\lib\tinyusb\src/class/audio/audio_device.h:70:2: error: #error You must define an audio class control request buffer size!
   70 | #error You must define an audio class control request buffer size!

sandeepmistry, 2021-07-05:
Hi @gigamegawatts,

Thanks for reporting this, I can confirm this project is not compatible with `TinyUSB 0.10.1` - however stay tuned for a pull request to make it compatible with the TinyUSB version that is used with v1.2.0 of the Pico SDK.

sandeepmistry, 2021-07-05:
@gigamegawatts please try out PR #2 if you get a chance and provide any feedback you have on it.

gigamegawatts, 2021-07-05:
Hi,

Yes, that fixed the problem.  It builds successfully, and the USB PDM Microphone example works on Linux.

Thanks for the quick response!

Dan

PR #2 (2021-07-05): Update usb_microphone example to be compatible with Pico SDK v1.2.0
Resolves #1.

cc/ @gigamegawatts
closed 2021-07-05 by gigamegawatts