ArmDeveloperEcosystem/microphone-library-for-pico #3: Support mic array?

nanfang2000, 2021-07-28:
Is it possible to support multiple PDM microphone and as USB microphone?

sandeepmistry, 2021-08-17:
Hi @nanfang2000,

This should be possible to do for stereo PDM microphone input, however it is currently not supported by this library.

sandeepmistry, 2021-10-05:
Happy to review a pull request for this feature. At this time there are no plans for our team to develop this feature.

CaydenPierce, 2022-03-10:
Hey @nanfang2000 , did you make any progress here?

@sandeepmistry any idea if we could go past two microphones, given that the microphone uses RP2040 PIO, which there are only two of?

CaydenPierce, 2022-03-15:
I have started this here: https://github.com/CaydenPierce/MSA

Modifying the system to support multiple microphones at once with a master PIO state machine for data and clock and slave PIO state machines for just data. Currently I have 4 microphones streaming data in.

@sandeepmistry my repo is a fork of a fork, but I would be up to PR this stuff if we can make it work, checkout the repo

StuartIanNaylor, 2022-10-02:
> I have done this here: https://github.com/CaydenPierce/MSA
> 
> Modifying the system to support multiple microphones at once with a master PIO state machine for data and clock and slave PIO state machines for just data. Currently I have 4 microphones streaming data in.
> 
> @sandeepmistry my repo is a fork of a fork, but I would be up to PR this stuff if we can make it work, checkout the repo

Caydan that is really cool you also managed 48k x4 ?
Anyone done this with I2S than PDM as notice the aliexpress modules seem to be much cheaper and also you could use ADC's.

This is really cool as respeaker do a collection of hats using TDM on the single pi port and there is no way to guarantee the channel order and in my mind totally useless even if sold by various vendors.

On the Pi I molested some code to do a delay-sum beamformer where really the only load is the GCC-PHAT to get TDOA.
https://github.com/StuartIanNaylor/ProjectEars/tree/main/ds
2 channels on a zero2 runs about 60% on a single core but never progressed as the respeaker hats and clones because of the lack of TDM sync are sort of garbage.
Also have fixed mics of a geometry ill suited for delay some due to aliasing makes them even more pointless.

Invensense did a great app not on simple beamforming https://invensense.tdk.com/download-pdf/an-1140-microphone-array-beamforming/

Where at 48Khz your 2 channel broadside should be approx 60-75mm which works in my above code molestation and if anyone is capable of providing a better neon optimised FFT then please do.
It occured to me you could make a 4 channel directional array with only the overhead of GCC-Phat of 2 mics as likely the 2nd rear broadside can use the same TDOA as the front and the 21mm spacing between those 2 is static and likely could be calculated via the angle of the TDOA than running GCC_Phat gain.
Never got round to implementing it due to the bad TDM/Fixed geometry implementation of the respeaker hats but the Pico/USB is sounding very interesting.
Also wonder if you could get another port for an ADC that could be hardware loopback channels even though only one needed.

longhd2, 2024-01-05:
> I have done this here: https://github.com/CaydenPierce/MSA
>
> Modifying the system to support multiple microphones at once with a master PIO state machine for data and clock and slave PIO state [...]

I have built according to your git.  But when copying to Pico it doesn't work.  Although when building for 1 mic pdm thep git in the main branch it still works.  Can you help me find out what's wrong?

delingren, 2024-06-18:
Thanks! I just tried your code with 2 mics and it worked. However, there seems to be a lot of high frequency hissing noise which doesn't exist with the original single channel code. Any idea what the issue might be? If it helps, I don't need a wide frequency range as I'm using my mics for video meetings (human voices).

> I have done this here: https://github.com/CaydenPierce/MSA
>
> Modifying the system to support multiple microphones at once with a master PIO state machine for data and clock and slave PIO state [...]

lukeinator42, 2024-08-09:
@CaydenPierce I am also interested in how to get this working. I built your firmware and wired my mics up similar to the pcb schematic (I changed mics 3 and 4 to GPIO 14 and 15), but the usb device doesn't show up. 

Did you need to overclock the pico? When I overclock mine, the device shows up but doesn't work as an audio device.

yoonmo-aynak, 2025-05-12:
@CaydenPierce , Hello thanks for your repo. So from my understanding your code can take multiple channel mic inputs through multiple wires which means that each data line from each mic is connected to a different GPIO am I right? The current mic module that I am using right now has two mics mounted inside the module and share the same data and clk lines. Do you think that is possible with the PICO SDK? I am pretty new to Raspberry pi and embedded systems so this might be a dumb question lol.

delingren, 2025-06-30:
@yoonmo-aynak I have forked this repo to do exactly that. It was hacky and not well documented. I also don't remember much about it. But here's the code if you want to take a look and get some ideas.
https://github.com/delingren/mems_mic/tree/stereo_no_encoding