ArmDeveloperEcosystem/microphone-library-for-pico #6: Device cannot start error on Windows 10

iovsiann, 2021-09-14:
Hi,
  thank you for the sample code. I tried it on Ubuntu 18.04 and Windows. While Ubuntu 18.04 finds MicNode and shows it in Audacity, the device fails to start in Windows 10.
- MicNode does appear under "Sound, video and game devices" in Device Manager, but with a yellow triangle icon (apparently indicating that the device failed to start)
- right-clicking on MicNode in Device Manager shows device status as "This device cannot start. (Code 10) The specified range could not be found in the range list."

Any advice  would be much appreciated.
Thank you

iovsiann, 2021-09-14:
Sorry, incorrect project

sandeepmistry, 2021-09-23:
Hi @iovsiann,

I'm having issues with my Windows 10 VM, the USB microphone device is listed in the device manager with the yellow triangle. However, I do NOT see it listed in Audacity.

A few suggestions to try while I try to figure out my setup would be:

1) Port an upstream patch over https://github.com/hathach/tinyusb/pull/1058 over, you'll need to change [examples/usb_microphone/usb_descriptors.c](https://github.com/ArmDeveloperEcosystem/microphone-library-for-pico/blob/main/examples/usb_microphone/usb_descriptors.c#L95):

replace:
```c
TUD_CONFIG_DESCRIPTOR(1, ITF_NUM_TOTAL, 0, CONFIG_TOTAL_LEN, TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP, 100),
```
with:
```c
TUD_CONFIG_DESCRIPTOR(1, ITF_NUM_TOTAL, 0, CONFIG_TOTAL_LEN, 0x00, 100),
```

2) Try to change Audacity to record at 16,000 Hz

3) Consider reaching out to the [TinyUSB team](https://github.com/hathach/tinyusb), as the USB microphone example in this project was based on the [TinyUSB audio_test device example](https://github.com/hathach/tinyusb/tree/master/examples/device/audio_test) - so they may be able to provide additional guidance or insights.

sandeepmistry, 2021-09-23:
@iovsiann I think I've tracked this down.

Please try out the changes proposed in https://github.com/ArmDeveloperEcosystem/microphone-library-for-pico/pull/8 and let me know how it goes.

iovsiann, 2021-09-24:
@sandeepmistry,
  I have checked it, it works both on my Windows laptop (Skype, Audacity) and Ubuntu desktop (Audacity) - I used your code in the pull request without any changes. Everything, including Audacity, worked "out of the box" - I didn't have to tweak Audacity mic sampling rate.
Many thanks for the fix!

PR #8 (2021-09-24): Fixes for microphone example on Windows
These changes should resolve #6.

They include:

1) a port of https://github.com/hathach/tinyusb/pull/1058

2) bumping `CFG_TUD_AUDIO_EP_SZ_IN` by one, as per the comment in: https://github.com/hathach/tinyusb/blob/2eaf99e0aa9c10d62dd8d0a4e765f5941bfeaf98/examples/device/audio_4_channel_mic/src/tusb_config.h#L104
```c
// 48 Samples (48 kHz) x 2 Bytes/Sample x CFG_TUD_AUDIO_N_CHANNELS_TX Channels
// - the Windows driver always needs an extra sample per channel of space more, otherwise it complains... found by trial and error

```

So far, I've tested on a [... cut]
closed 2021-09-24 by iovsiann