ArmDeveloperEcosystem/microphone-library-for-pico #7: Background low-frequency, fluctuating tone in recordings

iovsiann, 2021-09-18:
Hi,
  thank you for the sample code. Microphone sound capture works, but there it comes with background low-frequency tone that fluctuates (on/off a few times per second). I've attached a WAV recording - the fluctuating background tone is easy to hear at the end of the recording.
  Also, it appears that the audio has some breaks in it (i.e. sound gets missing momentarily).
[untitled.wav.zip](https://github.com/ArmDeveloperEcosystem/microphone-library-for-pico/files/7189242/untitled.wav.zip)
  Any help would be much appreciated.
Thank you

sandeepmistry, 2021-09-22:
Hi @iovsiann,

I've tried this USB microphone example again with an RPi Pico + Adafruit PDM mic. on a breadboard, then using Quicktime to record on macOS, and it seems ok here:

[hello.m4a.zip](https://github.com/ArmDeveloperEcosystem/microphone-library-for-pico/files/7213883/hello.m4a.zip)

Would you be able to share more details of your setup?
- which board and microphone + photo of wiring
- what OS and app you are using for the recording
- I think you are using the USB mic. example but please confirm

I'm wondering if the PDM mic. is configured for the wrong channel, for the `MP34DT01-M` I have, I've grounded the `L/R` (SEL) pin, so data is valid when the CLK is low. See section 6 of [MP34DT01-M.pdf](https://cdn-learn.adafruit.com/assets/assets/000/049/977/original/MP34DT01-M.pdf).

iovsiann, 2021-09-23:
@sandeepmistry,
  thank you for checking. It seems the problem is on my side. I was using VMPlayer running Ubuntu 18.04 to capture sound. When I plug the Pico in a real Ubuntu PC, the mic works. I also tried NVIDIA Nano running Ubuntu and it worked as well.

closed 2021-09-23 by iovsiann

sandeepmistry, 2021-09-23:
@iovsiann great to hear things are working! Thanks for diagnosing the issue and sharing the outcome!