ArmDeveloperEcosystem/microphone-library-for-pico #12: Use Micropython With microphone library

alirezadigi, 2022-04-26:
Hi. How can i use micropython firmware alongside a Max9814 , with this module?
Any Solutions?

I have written the code below but cant hear clear voice in audacity...
```python
from machine import Pin, ADC
import ustruct , time

analog_value = machine.ADC(26)
conversion_factor =3.3/(65536)

samples = []

while True:
    reading = analog_value.read_u16()*conversion_factor   
samples.append(int(reading)) #print("ADC: ",reading)
time.sleep(0.002)

with open('Voice.bin', 'wb') as output:
     for sample in samples:
        output.write(struct.pack('<h', sample))

```

alirezadigi, 2022-04-26:
Or even use this microphone library near MicroPython...

sandeepmistry, 2022-04-26:
Hi @alirezadigi,

You would have to create a [Native machine .mpy file](https://docs.micropython.org/en/latest/develop/natmod.html) which bridges this library to MicroPython.

This is out of scope of this project.

alirezadigi, 2022-04-27:
@sandeepmistry thanks 
sorry can you please help me with the python code? ...

closed 2022-07-03 by stale[bot]