ArmDeveloperEcosystem/microphone-library-for-pico #13: Reducing Gain of the Microphone

aryanvgithub, 2022-05-26:
Hi, I was trying to record some audio using the same microphone setup as you instructed. It was a loud audio and for the same reason clipping was observed when the recording was analyzed. I tried to change the "Filter Gain" and "Volume" parameters to reduce the gain but didn't work. It would be great to some lead on reducing the input gain.

sandeepmistry, 2022-05-30:
@aryanvgithub can you please provide more details on your setup:

- Exact hardware used
- What software was used to record the audio
- How close the audio source is to the microphone
- Values used to change the filter gain and volume parameters

an example recording would also be useful.

aryanvgithub, 2022-05-31:
The details are as follows:
1. The PDM mic along with the PICO was made as instructed in your tutorial. The whole system was then used as a USB microphone for a windows 11 Laptop.
2. I have used Python SoundDevice and Scipy  library to record audio as WAV file.
3. The audio source is next to the mic nearly a centimeter away.
4. The default Filter Gain was defined as 16, I tried to change it to 8.

I am working on a project that involves recording breathing sounds. So, the microphone is placed next to the nose in the experiment.

I have attached below the sound file as it is seen in Audacity to check Clipping, The peaks in the recording refers to the exhalation. 
[image]

I am not able to upload the Audio file as this platform does not support uploading .wav files. 

Thanks for replying promptly!!

sandeepmistry, 2022-06-09:
Hi @aryanvgithub,

> I am not able to upload the Audio file as this platform does not support uploading .wav files.

You can try to create a .zip file of the .wav file and upload.

Thanks for the info. I was able to reproduce what you were seeing above. Setting the filter gain to `1` in the USB microphone example removes the clipping for me:

```diff
diff --git a/examples/usb_microphone/main.c b/examples/usb_microphone/main.c
index 9323110..922aba1 100644
--- a/examples/usb_microphone/main.c
+++ b/examples/usb_microphone/main.c
@@ -38,6 +38,7 @@ int main(void)
   // initialize and start the PDM microphone
   pdm_microphone_init(&config);
   pdm_microphone_set_samples_ready_handler(on_pdm_samples_ready);
+  pdm_microphone_set_filter_gain(1);
   pdm_microphone_start();
 
   // initialize the USB microphone interface
```

I would suggest you start when a gain of `1` and then increase it as needed.

sandeepmistry, 2022-07-29:
Closing as resolved, please re-open as needed.

closed 2022-07-29 by sandeepmistry