ArmDeveloperEcosystem/microphone-library-for-pico #16: Changing the sample rate

jh2srv, 2022-06-17:
Hi,
I tried to change the sample rate without success (tested only in windows 10).
I only changed tusb_config.h to:
`#define CFG_TUD_AUDIO_EP_SZ_IN                                        (48+1) * CFG_TUD_AUDIO_FUNC_1_N_BYTES_PER_SAMPLE_TX * CFG_TUD_AUDIO_FUNC_1_N_CHANNELS_TX      // 16 Samples (16 kHz) x 2 Bytes/Sample x 1 Channel
`
and usb_microphone.h :
`#define SAMPLE_RATE 48000`

any ideas what else need to be modified?

sandeepmistry, 2022-07-25:
Hi @jh2srv,

> I tried to change the sample rate without success

Could you please provide more details on the unexpected behaviour you are seeing? Have you tried to use Wireshark to inspect the USB traffic?

jh2srv, 2022-07-31:
Hi @sandeepmistry,
It was working with 16k Hz and after changing  the sample rate to 48000 Hz didn't work anymore (see ny first post).
I fixed just uninstalling the sound driver and letting windows to reinstall the driver after pluging again the pico.
Now is working with 48000 Hz...

closed 2022-07-31 by jh2srv

sandeepmistry, 2022-08-02:
@jh2srv thank you for the update, it is good to know that you have it working and there is work around by reinstalling the driver.