ArmDeveloperEcosystem/microphone-library-for-pico #17: Sample rate lower than 16000 may cause some PDM microphone to stop outputting data

rxing365, 2022-07-23:
The output frequency of the clk pin is calculated by the sample rate, and some PDM microphone has a limited frequency range for clock speed (Like MP34Dxxx  has a clock rate limit from1 to 3.25 MHz), which causes the sample rate below 16000(Corresponding to the clk frequency of 1.024MHz) to sometimes fail to make the microphone work.

sandeepmistry, 2022-07-25:
Hi @rxing365,

How do you suggest this be addressed? Would you like to see a new API added?

Would changing the [`PDM_DECIMATION` macro](https://github.com/ArmDeveloperEcosystem/microphone-library-for-pico/blob/main/src/pdm_microphone.c#L21) to `128` work for your use case?

rxing365, 2022-07-26:
Really appreciate your reply. I tested it on my adafruit board and it works. I'm trying to run some voice recognition algorithms on pi pico, it doesn't need much sound quality, the high sample rate is a big challenge for memory. A sample rate of 8000Hz is a reasonable value for my audio buffer, but it would be more convenient to have an API that would allow me to set a lower value. The PDM to PCM conversion seems to be implemented with a lookup table, does that mean it would be hard to implement functions for arbitrary decimation values? It would be a cool feature if the library could automatically adjust the decimation settings to a range acceptable to the sensor, or manually set arbitrary decimation values.

ej159, 2025-02-25:
I was getting a constant "-512" output over serial for the example case with an MP34DT01. Upping the sample rate to 16000 fixed this (as did disconnecting the 3.3V pin to the board).