ArmDeveloperEcosystem/microphone-library-for-pico #20: What is the maximum sample-rate?

sail2themoon, 2022-09-01:
Assuming the PDM-mic is capaple of doing this: What is the maximum audio sample rate for the Pi pico 2040 and this library? Are 192KHz or even 384KHz possible?

Kind regards

sandeepmistry, 2022-10-03:
Hi @sail2themoon,

This is a great question. I've only tested 8 kHz and 16 kHz myself.

Please let me know if you've managed to get a higher sampling rate working.

sail2themoon, 2022-12-27:
the highest sample-rate I can get to work is 88KHz. The strange thing is: 80Khz does not work: loud crackling in the recording.
I've tried also 37KHz and 62KHz, just to have some other non-regular sample-rates. They are fine.
This is my mic: SPH0641LU4H-1

sail2themoon, 2022-12-27:
This is a spectrogram of the 99KHz recording clicking my fingers:
https://i.postimg.cc/kMVv8N2F/Bildschirmfoto-vom-2022-12-27-21-20-49.png
There is some silent hiss above frequencies humans can hear, and I don't know if that's a recording error or if that sound was some interference of a device here. But the recording sounds fine. You can also see, that there is data at those high frequencies. So the mic really gets something.

sail2themoon, 2023-04-14:
Hi RG,
I think the mic can to that well, as seen in the datasheet. But you have to sample with 80KHz then, for picking up 40Khz waves. The pdm library got me confused.. As I wrote some sample-frequencies go well, a slightly higher suddenly won't, an even higher works again, then again some which won't work... So trouble-free: I'd say no. I think first someone must find the right way to get data from it at high sample-rates.

VoiceMagician, 2024-06-07:
I have try 16kHz, 32kHz, 48kHz, 60kHz and 64kHz, they cant work except 64kHz.

PureLin, 2025-05-16:
Hi, a dumb question: how to change the sample rate?
I tried to change the marco SAMPLE_RATE and SAMPLE_BUFFER_SIZE in usb_microphone.h but it don't work, the USB input device will not show up with sample rate big than 18Khz.