ArmDeveloperEcosystem/microphone-library-for-pico #21: Can it work with?

Lanson2232, 2022-09-27:
So Actually I'm trying to create a "Sound card" for my Electric Guitar but it didn't work with [this Elektret microphone](https://www.robotistan.com/ses-sensor-karti-mikrofon-sensoru) aka. Sound sensör? isn't it compatible or do I need do something I would use PDM but this sort of microphone modules are much easier to replace or solder on it
I don't get any soundwave on audacity just a straight line

kannel-outis, 2022-10-24:
@Lanson2232 your microphone only gives an output of only 0s and 1s... 0 for no sound and 1 for sound... It's not compatible for what you want to use it for. It only senses sound

Lanson2232, 2022-11-06:
> @Lanson2232 your microphone only gives an output of only 0s and 1s... 0 for no sound and 1 for sound... It's not compatible for what you want to use it for. It only senses sound

Well that makes sense. Anyway thanks for helping both. I did create "Sound Card" or "Guitar mic" by making an basic AMP circuit that makes analog to analog to my computer

closed 2022-11-06 by Lanson2232