ArmDeveloperEcosystem/microphone-library-for-pico #22: Microphone does not work on Windows 10

chenjy96, 2023-02-09:
I appreciated your work, but I have several questions about this project.
1. Is it correct to change sample rate in [this line](https://github.com/ArmDeveloperEcosystem/microphone-library-for-pico/blob/main/examples/usb_microphone/tusb_config.h#L103) of code?(for example 16 for 16khz, 48 for 48 kHz) 
2. On my windows computer, the USB microphone only works when the sample rate is 16kHz, I tried many other parameters(32 40 48 80 etc.), but none of them work properly. (Although the microphone seem to be working with these parameters on Ubuntu.)
I check the error message from Audacity , and it is 

`
{
    "timestamp": 1675922013,
    "event_id": "de2968e644af2446aafc55d7a0425be4",
    "platform": "native",
    "release": "audacity@3.1.3",
    "contexts": {
        "os": {
            "type": "os",
            "name": "Windows",
            "version": "10.0.19044"
        }
    },
    "exception": {
        "values": [
            {
                "type": "Error",
                "value": "Error opening recording device.\nError code: -9999 Unanticipated host error.",
                "mechanism": {
                    "type": "runtime_error",
                    "handled": false
                }
            }
        ]
    }
}
`

I tried to reinstall sound driver mention in [this issue](https://github.com/ArmDeveloperEcosystem/microphone-library-for-pico/issues/16), but it does not work. Could you help me to solve this?

sandeepmistry, 2023-02-09:
Hi @chenjy96,

For 1) yes.

@jh2srv since you got this working in https://github.com/ArmDeveloperEcosystem/microphone-library-for-pico/issues/16#issuecomment-1200473135 - do you have any tips for @chenjy96 on how they can re-install the sound driver for 2) ?

jh2srv, 2023-02-28:
Hi, 
I don't have any tips for this issue. Maybe it is an error from audacity, what audio host are you using in audacity? If you choose WASAPI it will not work if you use a sample rate different to the one you defined in your code.

closed 2023-03-27 by chenjy96