ArmDeveloperEcosystem/microphone-library-for-pico #28: Using with Arduino

euphemism, 2023-09-25:
Hi Sandeep,

I have been in the midst of (and in over my head with) a project to record voice messages via a handheld, battery-powered device. I designed and built it around the RP2040 and the Adafruit PDM microphone breakout board. I have been attempting to get decent, functional audio out of it via CircuitPython and failing. I came across your USB microphone project and the audio generated sounds great.

I am not a C developer, nor do I have any experience with hardware. Is there an easy way to incorporate the PDM microphone library in this project with, say, an Arduino project? I don't need the USB functionality, just the ability to extract PCM samples from the microphone input.

sandeepmistry, 2023-09-26:
Hi @euphemism,

Arduino support is out of scope of this project, the Arduino team does have a PDM library that is compatible with the RP2040:
- https://github.com/arduino/ArduinoCore-mbed/tree/main/libraries/PDM/src
- https://github.com/arduino/ArduinoCore-mbed/tree/main/libraries/PDM/src/rp2040

euphemism, 2023-10-03:
Hi Sandeep,

Thanks for the reply. Ended up getting your library working with PlatformIO + Arduino via the [unofficial Pico/RP2040 core](https://arduino-pico.readthedocs.io/en/latest/).

closed 2023-12-10 by stale[bot]