ArmDeveloperEcosystem/microphone-library-for-pico #33: Error on cmake ,,   command

thedistinguishedlunatic, 2024-05-18:
I receive an error when compiling 👍 there is no Cmakelists.txt  for the source file

sandeepmistry, 2024-05-21:
Hi @thedistinguishedlunatic,

Could you please provide more information on the full set of steps you are using? Thanks.

thedistinguishedlunatic, 2024-05-23:
cmake .. -DPICO_BOARD=pico
PICO_SDK_PATH is /home/pi/pico-sdk
PICO platform is rp2040.
Build type is Release
PICO target board is pico.
Using board configuration from /home/pi/pico-sdk/src/boards/include/boards/pico.h
TinyUSB available at /home/pi/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /home/pi/pico-sdk/lib/btstack
cyw43-driver available at /home/pi/pico-sdk/lib/cyw43-driver
Pico W Bluetooth build support available.
lwIP available at /home/pi/pico-sdk/lib/lwip
mbedtls available at /home/pi/pico-sdk/lib/mbedtls
CMake Error at CMakeLists.txt:44 (add_subdirectory):
 add_subdirectory given source "examples/hello_analog_microphone" which is
 not an existing directory.
CMake Error at CMakeLists.txt:45 (add_subdirectory):
 add_subdirectory given source "examples/hello_pdm_microphone" which is not
 an existing directory.
CMake Error at CMakeLists.txt:46 (add_subdirectory):
 add_subdirectory given source "examples/usb_microphone" which is not an
 existing directory.
-- Configuring incomplete, errors occurred!
See also "/home/pi/build/CMakeFiles/CMakeOutput.log".

 

&mdash;