ArmDeveloperEcosystem/microphone-library-for-pico #36: Add support for Raspberry Pi Pico 2

kholia, 2024-09-28:
It would be awesome to add support for Raspberry Pi Pico 2.

Pico 2 has fixed the ADC bug which is relevant for https://github.com/batica81/microphone-library-for-pico work.

Thanks!

sandeepmistry, 2024-09-30:
Hi @kholia,

Have you tried the library as is on the RPi Pico 2?

kholia, 2024-09-30:
This library works fine as is! Thanks for the pointer.

Kernel logs:

```
$ sudo dmesg
[41238.417308] usb 3-1.1.4: new full-speed USB device number 10 using xhci_hcd
[41238.546696] usb 3-1.1.4: New USB device found, idVendor=2e8a, idProduct=000f, bcdDevice= 1.00
[41238.546703] usb 3-1.1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[41238.546707] usb 3-1.1.4: Product: RP2350 Boot
[41238.546709] usb 3-1.1.4: Manufacturer: Raspberry Pi
[41238.546712] usb 3-1.1.4: SerialNumber: E3EDBFC93E15EF37
[41238.692969] usb-storage 3-1.1.4:1.0: USB Mass Storage device detected
[41238.693143] scsi host0: usb-storage 3-1.1.4:1.0
[41238.693292] usbcore: registered new interface driver usb-storage
[41238.694881] usbcore: registered new interface driver uas
[41239.731099] scsi 0:0:0:0: Direct-Access     RPI      RP2350           1    PQ: 0 ANSI: 2
[41239.731482] sd 0:0:0:0: Attached scsi generic sg0 type 0
[41239.732349] sd 0:0:0:0: [sda] 262144 512-byte logical blocks: (134 MB/128 MiB)
[41239.733760] sd 0:0:0:0: [sda] Write Protect is off
[41239.733765] sd 0:0:0:0: [sda] Mode Sense: 03 00 00 00
[41239.735761] sd 0:0:0:0: [sda] No Caching mode page found
[41239.735766] sd 0:0:0:0: [sda] Assuming drive cache: write through
[41239.743864]  sda: sda1

$ cp ./examples/usb_microphone/usb_microphone.uf2 /media/user/RP2350

$ sudo dmesg
[41300.185804] usb 3-1.1.4: new full-speed USB device number 11 using xhci_hcd
[41300.309025] usb 3-1.1.4: New USB device found, idVendor=cafe, idProduct=4010, bcdDevice= 1.00
[41300.309032] usb 3-1.1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[41300.309035] usb 3-1.1.4: Product: MicNode
[41300.309037] usb 3-1.1.4: Manufacturer: PaniRCorp
[41300.309038] usb 3-1.1.4: SerialNumber: 123456
```

I am closing this issue now - thanks!

closed 2024-09-30 by kholia